eclipse-threadx/threadx #424: ThreadX Queue Message Size Limit

lcdevans, 2024-12-03:
Hello,

I'm using ThreadX on an STM32 platform. I'm confused about the queue message size limit of 64 bytes/16 words.

I have read a previous Issue ticket #122 where the user asked about it, and was directed to another link for an explanation (https://learn.microsoft.com/en-us/answers/questions/292596/about-the-limite-of-the-message-size-of-queue). I have read both threads and am still confused about this.

The reason that seems to be given in the links above is that copies of larger sizes are too inefficient. Surely that should be up to the user's discretion? If a user wants to have a larger size and is aware of any inefficiencies involved with copying data, why does the ThreadX platform need to artificially restrict them? The answers given to #122 seem to indicate there is no technical reason the maximum message size can't simply be bigger. 

I understand there are workarounds, such as redefining `TX_16_ULONG`, editing the source file, disabling error checking, sending multiple messages, etc, but it is frustrating to have to add such complications to a project to get around what appears to be an arbitrary restriction. Similar platforms, such as FreeRTOS, have a much higher queue message size limit.

Can you provide any information about _why_ this limit of 64 bytes exists? Is there a technical reason other than copy size performance? Could you consider allowing larger sizes out-of-the-box?

Thanks for your time.

fdesbiens, 2024-12-05:
Hi @lcdevans.

Thank you for your question. I will find someone on the team who could provide context around this limit. Stay tuned.

fdesbiens, 2024-12-05:
@rahmanih Please feel free to comment.

rahmanih, 2024-12-20:
> One technical reason for this limit is that ThreadX messages are passed by value, i.e., copied from source to destination. The copy occurs with interrupts disabled, so the limit is 16 32-bit words. An application that doesn't care about the additional interrupt latency can override the TX_16_ULONG constant or directly all the internal _tx_queue_create API that does not have the error checking.

we may add a flag like ````TX_QUEUE_DISABLE_MSG_SIZE_CHECKING```` undefined by default, and use it in the <code>txe_queue_create()</code>?

````
#ifndef TX_QUEUE_DISABLE_MSG_SIZE_CHECKING
        /* Check for an invalid message size - greater than 16.  */
        else if (message_size > TX_16_ULONG)
        {

            /* Invalid message size specified.  */
            status =  TX_SIZE_ERROR;
        }
#endif
````

regards
Haithem.

billlamiework, 2024-12-20:
Yes, Haithem, this is a good approach that provides the desired flexibility!

lcdevans, 2024-12-30:
Thank you for the replies, a flag like that is what I was hoping would be possible.

fdesbiens, 2025-02-27:
@rahmanih Would you be willing to submit a PR implementing this feature?

Alternatively, anyone in @eclipse-threadx/iot-threadx-contributors interested? Let me know and I will assign the issue to you.

rahmanih, 2025-03-03:
I'll be working on that.

rahmanih, 2025-03-11:
> > One technical reason for this limit is that ThreadX messages are passed by value, i.e., copied from source to destination. The copy occurs with interrupts disabled, so the limit is 16 32-bit words. An application that doesn't care about the additional interrupt latency can override the TX_16_ULONG constant or directly all the internal _tx_queue_create API that does not have the error checking.
> 
> we may add a flag like `TX_QUEUE_DISABLE_MSG_SIZE_CHECKING` undefined by default, and use it in the `txe_queue_create()`?
> 
> ```
> #ifndef TX_QUEUE_DISABLE_MSG_SIZE_CHECKING
>         /* Check for an invalid message size - greater than 16.  */
>         else if (message_size > TX_16_ULONG)
>         {
> 
>             /* Invalid message size specified.  */
>             status =  TX_SIZE_ERROR;
>         }
> #endif
> ```
> 
> regards Haithem.

I thin we should keep the check on the message_size in the txe_queue_create() file.
thus I'm proposing the following implmentation
```
diff --git a/common/inc/tx_api.h b/common/inc/tx_api.h

index 3ff05f4b..8ed5fb63 100644

--- a/common/inc/tx_api.h

+++ b/common/inc/tx_api.h

@@ -328,6 +328,9 @@ extern   "C" {

 #define TX_TIMER_TICKS_PER_SECOND       (100UL)

 #endif

 

+#ifndef TX_QUEUE_MESSAGE_MAX_SIZE

+#define TX_QUEUE_MESSAGE_MAX_SIZE           TX_16_ULONG

+#endif

 

 /* Event numbers 0 through 4095 are reserved by Azure RTOS. Specific event assignments are:

 

diff --git a/common/src/txe_queue_create.c b/common/src/txe_queue_create.c

index 68d3540a..2d91cc9f 100644

--- a/common/src/txe_queue_create.c

+++ b/common/src/txe_queue_create.c

@@ -178,8 +178,8 @@ TX_THREAD       *thread_ptr;

             status =  TX_SIZE_ERROR;

         }

 

-        /* Check for an invalid message size - greater than 16.  */

-        else if (message_size > TX_16_ULONG)

+        /* Check for an invalid message size - greater than TX_QUEUE_MESSAGE_MAX_SIZE 16 by default.  */

+        else if (message_size > TX_QUEUE_MESSAGE_MAX_SIZE)

         {

 

             /* Invalid message size specified.  */

```

itf it o.k for @eclipse-threadx/iot-threadx-contributors I'll upload the PR (still to finalize the regression_tests).

regards.
Haithem

billlamiework, 2025-03-11:
This looks good to me!   Many thanks, Bill

closed 2025-04-15 by fdesbiens