eclipse-threadx/threadx #430: Problem with timeSetEvent using win32 port

Francis900, 2025-01-10:
I’m encountering two issues while trying to compile a project that contains ThreadX (master branch) in a Win32 port using Visual Studio Code on Windows 10:

Pointer type mismatch with timeSetEvent:
There’s an incompatible pointer type error when passing the third argument to timeSetEvent. The error message is:

Passing argument 3 of 'timeSetEvent' from incompatible pointer type [-Wincompatible-pointer-types].
It seems there’s an issue with the type of the third argument being passed to timeSetEvent.

Warning about #pragma comment:
I’m getting a warning:
#pragma comment: ignoring '#pragma comment' [-Wunknown-pragmas].
The line in question is:
#pragma comment (lib, "Winmm.lib").
This seems to be related to linking the Winmm.lib library, but even after manually adding the library paths to the compiler, it still can't find winmm.lib. I’m unsure why the pragma is being ignored.

I’m using the latest master branch of ThreadX, and I’m working in a Win32 environment with Visual Studio Code with gcc.

Any help with resolving these issues would be greatly appreciated.

```
\threadx\ports\win32\vs_2019\src\tx_initialize_low_level.c: In function '_tx_initialize_start_interrupts':
C:...\threadx\ports\win32\vs_2019\src\tx_initialize_low_level.c:272:69: error: passing argument 3 of 'timeSetEvent' from incompatible pointer type [-Wincompatible-pointer-types]
  272 |     _tx_win32_timer_id = timeSetEvent(TX_TIMER_PERIODIC, wTimerRes, _tx_win32_timer_interrupt, 0, TIME_PERIODIC);
      |                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                     |
      |                                                                     void (*)(UINT,  UINT,  DWORD,  DWORD,  DWORD) {aka void (*)(unsigned int,  unsigned int,  long unsigned int,  long unsigned int,  long unsigned int)}
In file included from C:/msys64/mingw64/include/mmsystem.h:27,
                 from C:/msys64/mingw64/include/windows.h:86,
                 from C:.../threadx/ports/win32/vs_2019/inc/tx_port.h:162,
                 from C:.../threadx/common/inc/tx_api.h:133,
                 from C:...\threadx\ports\win32\vs_2019\src\tx_initialize_low_level.c:28:
C:/msys64/mingw64/include/mmiscapi2.h:32:85: note: expected 'LPTIMECALLBACK' {aka 'void (*)(unsigned int,  unsigned int,  long long unsigned int,  long long unsigned int,  long long unsigned int)'} but argument is of type 'void (*)(UINT,  UINT,  DWORD,  DWORD,  DWORD)' {aka 'void (*)(unsigned int,  unsigned int,  long unsigned int,  long unsigned int,  long unsigned int)'}
   32 | WINMMAPI MMRESULT WINAPI timeSetEvent(UINT uDelay, UINT uResolution, LPTIMECALLBACK fptc, DWORD_PTR dwUser, UINT fuEvent);
      |                                                                      ~~~~~~~~~~~~~~~^~~~

```

Jasinsky, 2025-01-21:
You are using mingw64 - ThreadX port is crafted for 32bit Windows environments.
Quickest resolution is to use 32bit mingw.

Jasinsky, 2025-01-21:
Also - compiling with GCC will not understand VS compiler related pragmas (i.e. linking winmm.lib by using #pragma).

fdesbiens, 2025-01-22:
Thank you for those comments, @Jasinsky. 

@Francis900 Did you try to compile in a 32-bit environment as suggested?

fdesbiens, 2026-09-10:
Thank you for the report, @Francis900, and thank you for the accurate diagnosis, @Jasinsky.

This is not a defect in the Win32 port. Both errors come from pointing a 64-bit toolchain at a port that is written for 32-bit Windows. `timeSetEvent` takes an `LPTIMECALLBACK`, whose last three parameters are `DWORD_PTR`. On a 32-bit target `DWORD_PTR` and `DWORD` are the same type, so the port's callback signature matches; on a 64-bit target `DWORD_PTR` becomes 64 bits wide and the signatures no longer agree. That is exactly what MinGW-w64 is reporting.

The good news is that things have moved on since you filed this. There is now a dedicated 64-bit simulator port in `ports/win64/vs_2022`, and it declares the callback with `DWORD_PTR` as the API requires. If you want a 64-bit build, please use that port instead of `ports/win32/vs_2019`. If you would rather stay on the Win32 port, build it with a 32-bit toolchain, for example the `mingw32` environment of MSYS2 rather than `mingw64`.

The second message is only a warning and is expected. `#pragma comment (lib, "Winmm.lib")` is a Microsoft compiler extension, so GCC ignores it and tells you so. Under MinGW you have to name the library on the link line yourself by adding `-lwinmm`. That applies to the Win64 port as well, since it uses `timeBeginPeriod` from the same library.

I am closing this, but please do reopen or comment if the Win64 port does not work for you under MinGW-w64.

closed 2026-09-10 by fdesbiens