eclipse-threadx/threadx #431: ThreadX modules support on Cortex-M33

Jasinsky, 2025-01-24:
I am running ThreadX from tag v6.4.1_rel with modules-enabled kernel.

When compiled with TX_SINGLE_MODE_SECURE - it hard faults after first enter to the scheduler and PendSV interrupt.
PendSV interrupt executes until returning from exception. 
EXC_RETURN was set to 0xFFFFFFBC and that was causing aforementioned hardfault.

After applying patch below - scheduler runs successfully.

```
diff --git a/ports_module/cortex_m33/gnu/module_manager/src/tx_thread_schedule.S b/ports_module/cortex_m33/gnu/module_manager/src/tx_thread_schedule.S
index e392380b..ba8c5fea 100644
--- a/ports_module/cortex_m33/gnu/module_manager/src/tx_thread_schedule.S
+++ b/ports_module/cortex_m33/gnu/module_manager/src/tx_thread_schedule.S
@@ -471,6 +471,12 @@ _skip_vfp_restore:
     LDMIA   r12!, {r4-r11}                          // Recover thread's registers
     MSR     PSP, r12                                // Setup the thread's stack pointer
 
+#ifdef TX_SINGLE_MODE_SECURE
+    LDR     r0, =0xFFFFFFFD                         // Exception return to secure
+#else
+    LDR     r0, =0xFFFFFFBC                         // Exception return to non-secure
+#endif
+    MOV     lr, r0
     BX      lr                                      // Return to thread!
```

It might be that I am not understanding Trustzone secure/non-secure implications.

Following addition was present in MemManage handler e.g. - other interrupts do not have it. 

Could someone suggest if this is a scheduler bug for CM33/CM23 and other CMSE architectures or I am configuring my microcontroller wrongly?

fdesbiens, 2025-01-24:
Thank you for this question, @Jasinsky. 

@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: can one of you please comment on this issue?

fdesbiens, 2026-09-23:
I confirmed that this was a bug in `v6.4.1_rel`. The GNU Cortex-M33 module manager’s `tx_thread_stack_build.s` contained the `TX_SINGLE_MODE_SECURE` conditional, but GAS did not preprocess the lowercase `.s` file. Both branches were therefore assembled, and `0xFFFFFFBC` overwrote the secure `0xFFFFFFFD` value in the initial thread frame.

I fixed this on `dev` in #672 by renaming the affected assembly source to `.S`, and I added checks that assemble the GNU module ports with their feature macros enabled. I do not recommend overriding `lr` in the scheduler because it must retain the saved `EXC_RETURN` properties, including the floating-point frame state.

I am closing this as fixed. Thank you for reporting it and for providing the diagnosis.

closed 2026-09-23 by fdesbiens