eclipse-threadx/threadx #434: Timer Interrupt Issue on Cortex-A53 with ThreadX on TI-AM64X

cliew-inno, 2025-01-31:
Platform & Configuration: TI-AM64X
Processor: Cortex-A53
ThreadX Version: 6.4.1
Timer Used: TI's peripheral timer - TIMER6

**Issue Description:**
Our team is currently porting ThreadX to both the Cortex-M4 and Cortex-A53 cores of the TI AM64X platform. We have successfully completed the porting process on the Cortex-M4, where we configured SysTick as the system timer without issues.
We are basically printing "hello world" with tx_thread_sleep(xxx) where xxx is # of ticks.
For the Cortex-A53, given that it follows the ARMv8 architecture, we have configured a peripheral timer provided by TI to generate periodic interrupts and call _tx_timer_interrupt(). The timer initialization appears to be correct, as we observe that the peripheral timer's IRQ is triggered as expected. However, we are facing an issue where _tx_timer_interrupt() does not behave as expected, leading to the following problems:

- If _tx_timer_interrupt() is not called, the peripheral timer continues to generate interrupts normally.
- Once _tx_timer_interrupt() is called, tx_thread_sleep(1000) does not function correctly—no timer interrupts are generated afterward.
- We verified the DAIF mask register, and all bits remain 0x0, confirming that interrupts are enabled at the CPU level.

We are stuck at _tx_thread_schedule()'s __tx_thread_schedule_loop() because timer interrupt is not generated.
( see pic attached)

[image]

Could you please advise on:
- any specific ThreadX configurations required to properly handle the system tick via a peripheral timer on Cortex-A53?
- debugging tip on ThreadX timer interrupt?

Any guidance would be greatly appreciated

cz172638, 2025-02-03:
adding @shiva-ti who has fingers in TI's fork

shiva-ti, 2025-02-06:
Hi @cliew-inno
We have the working port of Cortex-A53 on ThreadX for AM62A here:
https://github.com/TexasInstruments/mcupsdk-core-k3/tree/am62a_threadx
You can find the hello_world examples for A53 running on ThreadX also:
https://github.com/TexasInstruments/mcupsdk-core-k3/tree/am62a_threadx/examples/kernel/threadx/hello_world

The above should give you a reference to compare and debug, it's hard to point to the exact issue from your shared observations

cliew-inno, 2025-02-06:
Thank you so much for the reference! I was able to figure out the root cause, but I will refer to the link you provided to resolve in more sophisticated method!!

closed 2025-02-06 by cliew-inno