eclipse-threadx/threadx #435: Illegal Instruction: tx_initialize_low_level:

sathya-1207, 2025-02-03:
Describe the bug
When building ThreadX for ARM Cortex-A34 in AMP mode using the GNU toolchain, the tx_initialize_low_level file contains the following sequence to save the first available memory address:

tx_initialize_low_level.S 

The issue occurs at the instruction:

assembly
Copy
Edit
    LDR     x1, [x1]   // Dereferencing __top_of_ram
This dereferences __top_of_ram, which is illegal and results in _tx_initialize_unused_memory being updated with a garbage value.

After commenting out this block, ThreadX runs successfully, and I can create threads without issues.

Target Device
ARM Cortex-A34
ThreadX Version
[Specify the exact version]
Toolchain and Environment
Toolchain: GNU GCC (arm-none-eabi-gcc or aarch64-none-elf-gcc [Specify version])
Operating System: [Linux/Windows]
To Reproduce
Steps to reproduce the behavior:

Build ThreadX for Cortex-A34 in AMP mode using the GNU toolchain.
Run the system and observe _tx_initialize_unused_memory.
The value is garbage due to the illegal dereference of __top_of_ram.
Commenting out this block allows ThreadX to function correctly.
Expected Behavior
_tx_initialize_unused_memory should be assigned a valid address without dereferencing __top_of_ram, ensuring correct memory initialization.

Impact
Showstopper: Prevents proper initialization and causes unpredictable behavior in ThreadX.
Logs and Console Output
(If available, attach logs or debugging output here.)

Additional Context
This issue seems specific to the GNU toolchain. It might work as intended with IAR or ARM Compiler.
__top_of_ram should likely be treated as an absolute address rather than dereferencing it.

fdesbiens, 2025-02-04:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors. Please review this issue and provide feedback. Thanks!

fdesbiens, 2025-02-13:
A kind reminder to the members of @eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors to provide feedback. Thanks!

sathya-1207, 2025-03-28:
Follow-up on _tx_initialize_unused_memory Issue in Cortex-A34 AMP Mode

Hi Team,

I wanted to follow up on this issue, as I haven’t received any updates yet. To summarize:

The LDR x1, [x1] instruction in tx_initialize_low_level.S incorrectly dereferences __top_of_ram, leading to _tx_initialize_unused_memory being assigned a garbage value.

Commenting out this instruction allows ThreadX to function correctly.

For reference, here’s how the same logic is implemented in the **Cortex-R5 port,** which does not exhibit this issue:

[image]

Unlike the Cortex-A34 implementation, this version does not attempt to dereference __top_of_ram, ensuring correct memory initialization.

I’d appreciate any insights on whether this behavior is intentional or if a fix is planned. **If needed, I can submit a pull request with a proposed fix.**

Looking forward to your feedback.

Thanks,
Sathya Moorthy K

fdesbiens, 2026-09-10:
Thanks for the detailed report, @sathya-1207, and my apologies for the long delay. You are right, and your analysis is correct.

`__top_of_ram` is defined in the linker script as `__top_of_ram = .;`, which makes it a linker-defined symbol whose address is the value we want. `LDR x1, =__top_of_ram` already puts the top of RAM in `x1`, so the `LDR x1, [x1]` that follows reads whatever happens to be stored at that address and hands the garbage to `_tx_initialize_unused_memory`. The startup code in the same example builds gets this right with `ldr x5, =__top_of_ram` and no dereference.

The Arm Compiler ports are not affected because they load a literal pool entry declared as `.quad (Image$$TOP_OF_RAM$$Base)`, where the dereference is correct. The GNU ports seem to have inherited the sequence without adjusting it. The SMP GNU ports already have the correct form.

I have opened #726 to remove the instruction from the 13 non-SMP GNU Cortex-A ports, the ARMv8-A architecture source they are generated from, and the two Cortex-A35 module example builds, which is every remaining occurrence in the tree. Thank you again for taking the time to file this and to follow up.

fdesbiens, 2026-09-23:
#726 removed the extra dereference from the affected GNU Cortex-A ports and merged into `dev`. The fix has not reached `master` yet. I am closing this issue as fixed on `dev`; the pull request reports passing port checks and no hardware run.

closed 2026-09-23 by fdesbiens