eclipse-threadx/threadx #437: Calling threadX APIs while interrupts are disabled

ashwinmalaviya, 2025-02-11:
We would like to understand  what could be issue if we call below APIs when global intrupts are disabled. 

tx_semaphore_get() //with infinite timeout
tx_event_flags_info_get and tx_event_flags_set
tx_event_flags_get()/ with infinite timeout

as far as I know, other RTOS do not allow the same(they don't allow the OS calls when interrupts are disabled) but in threadx I don't find any documentation which talks about calling threadx when Int are disabled 

it would be good if someone from the community provide any guidance.

thanks a lot..!!

wflynn, 2025-02-13:
To repeat what you are saying:
1) You disable global interrupts.
2) You try to get a mutex without a timeout. 
3) You want to know what might go wrong.  

Is this correct?

billlamiework, 2025-02-13:
You can call the ThreadX APIs with interrupts locked out. That is not a problem. The only issue is at the application level since you are increasing the time interrupts are locked out.

ashwinmalaviya, 2025-02-20:
> To repeat what you are saying:
> 
> 1. You disable global interrupts.
> 2. You try to get a mutex without a timeout.
> 3. You want to know what might go wrong.
> 
> Is this correct?

Not really but just wanted to understand if it is allowed to call these APIs within disable/enable all int block. other RTOS does not allow insert assert. i am not sure if threadx has some qual configuration

Vishnuchander90, 2025-02-21:
> You can call the ThreadX APIs with interrupts locked out. That is not a problem. The only issue is at the application level since you are increasing the time interrupts are locked out.

This is interesting question to understand from ThreadX. Lets say application disables all interrupts and calls ThreadX api then re-enables all interrupts, Does ThreadX might re-enable all interrupts again by end of it's API execution before application re-enables it?

billlamiework, 2025-02-21:
ThreadX will restore the interrupt posture from which the API was called. For example, if interrupts are disabled before a ThreadX API is called, the API will return with interrupts disabled. If interrupts are enabled before an API is called, they will be enabled upon return.

Vishnuchander90, 2025-02-24:
Thanks for the response.

Vishnuchander90, 2025-02-24:
> > To repeat what you are saying:
> > 
> > 1. You disable global interrupts.
> > 2. You try to get a mutex without a timeout.
> > 3. You want to know what might go wrong.
> > 
> > Is this correct?
> 
> Not really but just wanted to understand if it is allowed to call these APIs within disable/enable all interrupt block. Other RTOS does not allow and insert assert. I am not sure if Threadx has some equal configuration

Any updates on this?

billlamiework, 2025-05-13:
You can call ThreadX APIs with interrupts locked out, including APIs that block.  If the thread is suspended, the next highest priority thread is resumed with the interrupt posture of its last execution. If no other threads are ready, interrupts are enabled while ThreadX waits to schedule the next thread for execution.

fdesbiens, 2026-01-29:
Closing this issue since no fix was needed.