eclipse-threadx/threadx #443: ARC_EM threadx wrapped interrupts between TX_INITIALIZE_IN_PROGRESS to scheduler

amgross, 2025-02-27:
**Describe the bug**
In the ARC_EM port readme, there is no mention of limitations on running threadx rapped ISR before sheduler running:
https://github.com/eclipse-threadx/threadx/blob/06dabb0ad0403f59889ce85eeb3d48dbd9d3759b/ports/arc_em/metaware/readme_threadx.txt#L133-L169
While from my tests most of time it is OK, as context save/restore will think it is nested interrupt and won't run special flow except of registers storing/restoring, this is because _tx_thread_system_state is not zero (TX_INITIALIZE_IN_PROGRESS/TX_INITIALIZE_ALMOST_DONE). But if the interrupt comes after setting _tx_thread_system_state to TX_INITIALIZE_IS_FINISHED while timer initialization done at TX_PORT_SPECIFIC_PRE_SCHEDULER_INITIALIZATION. _tx_thread_context_restore will assume idle state was interrupted and the interrupt will be returned to scheduler, skipping on what left in TX_PORT_SPECIFIC_PRE_SCHEDULER_INITIALIZATION (timers for example may be still disabled so no OS tick will occur).
https://github.com/eclipse-threadx/threadx/blob/06dabb0ad0403f59889ce85eeb3d48dbd9d3759b/common/src/tx_initialize_kernel_enter.c#L148-L159

**To Reproduce**
make interrupt in this point and see it returns to scheduler

**Expected behavior**
Documentation update or somehow fix

**Impact**
Took lots of debug time

**Logs and console output**
If applicable, add console logs or other types of debug information like Wireshark capture as `.zip` file.

**Additional context**
Add any other context about the problem here.

fdesbiens, 2025-02-27:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: anyone in the group with experience with that specific platform?

amgross, 2025-02-27:
Just want to mention this issue may be not just ARC_EM related. I referenced ARC_EM because it is the system I worked on and found it.

amgross, 2025-03-04:
My fix suggestion is to move `TX_PORT_SPECIFIC_PRE_SCHEDULER_INITIALIZATION ` to be before setting `_tx_thread_system_state =  TX_INITIALIZE_IS_FINISHED;` (which is anyway making sense because if we are running initialization then initiallization didn't finished...)

fdesbiens, 2025-03-04:
I will bring this topic for discussion at our weekly team meeting.

fdesbiens, 2025-03-19:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: Please  provide your feedback.

billlamiework, 2025-03-19:
In general, ThreadX assumes that no ThreadX-managed interrupts occur before ThreadX is initialized or during ThreadX initialization. This is true for almost all versions of ThreadX and is generally not a problem.  If interrupts must be fielded before ThreadX completes its initialization, some additional logic in either the ISR must be added to account for this... and such logic would amount to unnecessary overhead after initialization is complete.

fdesbiens, 2025-04-01:
Closing for now, as @billlamiework provided appropriate guidance. Feel free to reopen.

closed 2025-04-01 by fdesbiens