eclipse-threadx/threadx #444: Incorrect constant used in comparison.

hstokes1, 2025-03-10:
https://github.com/eclipse-threadx/threadx/blob/7ad78c40e9702917264a9b722890110fdb909ebc/ports/cortex_m4/iar/src/tx_iar.c#L289

Should be comparing with _MAX_FLOCK, not _MAX_LOCK.

Result of bug: 

When _MAX_FLOCK is greater than _MAX_LOCK, some file mutexes are incorrectly set to NULL, and the app will hard fault as soon as the file lock is encountered.

hstokes1, 2025-03-10:
Also lines 273, 681, and 698.

fdesbiens, 2025-03-21:
Can one of the @eclipse-threadx/iot-threadx-contributors have a look and provide feedback?

fdesbiens, 2026-09-03:
Thank you for the report and for the follow-up pointing at the additional lines, @hstokes1. Confirmed, and thanks to @billlamiework for the review.

Both `__iar_file_Mtxinit()` variants compare the file lock index against `_MAX_LOCK` while the array they index is `__tx_iar_file_lock_mutexes[_MAX_FLOCK]`, so the bound is simply wrong in all four places you identified.

The same code is duplicated in all 27 copies of `tx_iar.c` in the tree, so the real scope is 108 occurrences rather than four. PR #695 corrects all of them.

I will close this issue once the PR is merged.

fdesbiens, 2026-09-03:
PR #695 has been merged into `dev`. All 108 occurrences of the incorrect `_MAX_LOCK` bound across the 27 copies of `tx_iar.c` now use `_MAX_FLOCK`.
Thank you for the report! Closing this one.

closed 2026-09-03 by fdesbiens