eclipse-threadx/threadx #448: Newlib + threadx

Mula-Gabriel, 2025-04-12:
I use ThreadX with Newlib (or Newlib-nano) on several NXP microcontrollers.
I'm using the version of Newlib provided by NXP with MCUXpresso (version 4.3.0).
Newlib is built with multithreading support enabled (__SINGLE_THREAD__ is not defined) and with _RETARGETABLE_LOCKING enabled.

I’ve implemented all the __retarget_lock_*****  functions using ThreadX mutexes, and I’ve made sure in the linker script that the Newlib-managed heap doesn’t overlap with any byte pool or extended objects managed by ThreadX.
I chose this approach because it felt more "complete" than implementing only the specific lock hooks like __malloc_lock or __env_lock.
With this setup, I can use malloc or printf from multiple threads, and everything seems stable.

However, after reading [this section on reentrancy](https://sourceware.org/newlib/libc.html#Reentrancy), I’m concerned that I may not be managing __reent properly.

Option 1: Manually managing the struct _reent per thread seems error-prone to me.
Ideally, I’d like this structure to be handled automatically.

On FreeRTOS, there’s a compilation flag configUSE_NEWLIB_REENTRANT that updates _impure_ptr during vTaskSwitchContext.

Question: On ThreadX, what are the recommended practices to properly manage the _reent struct?

fdesbiens, 2025-04-15:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: Please provide feedback to the issue's author.

Mula-Gabriel, 2025-04-15:
After further investigation, I identified a potentially suitable solution by leveraging the **USER_EXTENSION** mechanism.

1) To detect thread context switches, I compiled ThreadX with the **TX_ENABLE_EXECUTION_CHANGE_NOTIFY** option enabled.
(On my target, this flag should be add as assembler additional option, because those fonction are called in **tx_thread_schedule.S**)

I define the following fonction somewhere in my code  :
```
void _tx_execution_thread_enter(void)
{
    extern TX_THREAD *     _tx_thread_current_ptr;
    _impure_ptr = &_tx_thread_current_ptr->newlib;
}

void _tx_execution_thread_exit(void)
{
	//Nothing special to do
}
```

2) To add a reent struct per thread i set (in **tx_user.h**)

`#define TX_THREAD_USER_EXTENSION  struct _reent newlib;`

and define init macro like that :
`#define TX_THREAD_CREATE_EXTENSION(thread_ptr) _REENT_INIT_PTR(&thread_ptr->newlib);`

Note that tx_port.h need a little change to no redefine TX_THREAD_CREATE_EXTENSION to empty : 

from
```
#define TX_THREAD_CREATE_EXTENSION(thread_ptr)
#define TX_THREAD_DELETE_EXTENSION(thread_ptr)
```

to
```
//may be defined in tx_user.h
#if !defined(TX_THREAD_CREATE_EXTENSION)
#define TX_THREAD_CREATE_EXTENSION(thread_ptr)
#endif

#if !defined(TX_THREAD_DELETE_EXTENSION)
#define TX_THREAD_DELETE_EXTENSION(thread_ptr)
#endif
```

What are your thoughts on this approach ?

fdesbiens, 2025-04-22:
I think @sanderson-AstroFirmwareSolutions mentioned that he had experience with Newlib. Can you confirm, Seth?

Mula-Gabriel, 2025-04-25:
I have no further questions, thank you. However, since newlib is widely used, I suggest adding this response to the documentation. Thank you again for your support.

fdesbiens, 2025-04-29:
Thank you for confirming, @Mula-Gabriel.

@sanderson-AstroFirmwareSolutions: Can I ask you to propose a way to add this information to our official documentation, please? You will find the source here: **https://github.com/eclipse-threadx/rtos-docs-asciidoc**

amgross, 2025-07-24:
@Mula-Gabriel can you please share your code for the __retarget_lock*****_ and the malloc/env locks?

Mula-Gabriel, 2025-07-24:
I have just define the lock as TX_MUTEX
```
#ifndef __SINGLE_THREAD__

struct __lock
{
	TX_MUTEX Mutex;
};
```

I add init_retarget_locks just to make thing compile, it's not called actually.
```
extern TX_THREAD *     _tx_thread_current_ptr;

ULONG tx_execution_isr_nest_counter = 0 ;

struct __lock __lock___sinit_recursive_mutex;
struct __lock __lock___sfp_recursive_mutex;
struct __lock __lock___atexit_recursive_mutex;
struct __lock __lock___at_quick_exit_mutex;
struct __lock __lock___malloc_recursive_mutex;
struct __lock __lock___env_recursive_mutex;
struct __lock __lock___tz_mutex;
struct __lock __lock___dd_hash_mutex;
struct __lock __lock___arc4random_mutex;

__attribute__((constructor)) static void init_retarget_locks(void)
{
    tx_mutex_create(&__lock___sinit_recursive_mutex.Mutex,"init",TX_INHERIT);
    tx_mutex_create(&__lock___sfp_recursive_mutex.Mutex,"sfp",TX_INHERIT);
    tx_mutex_create(&__lock___atexit_recursive_mutex.Mutex,"atexit",TX_INHERIT);
    tx_mutex_create(&__lock___at_quick_exit_mutex.Mutex,"aqe",TX_INHERIT);
    tx_mutex_create(&__lock___malloc_recursive_mutex.Mutex,"malloc",TX_INHERIT);
    tx_mutex_create(&__lock___env_recursive_mutex.Mutex,"env",TX_INHERIT);
    tx_mutex_create(&__lock___tz_mutex.Mutex,"tz",TX_INHERIT);
    tx_mutex_create(&__lock___dd_hash_mutex.Mutex,"dd",TX_INHERIT);
    tx_mutex_create(&__lock___arc4random_mutex.Mutex,"arc4r",TX_INHERIT);
}
```

All recursive function are called :

```
void __retarget_lock_init_recursive(_LOCK_T *lock)
{
    *lock = calloc ( 1,sizeof(struct __lock) );
    tx_mutex_create(&(*lock)->Mutex,"newlib_rlock",TX_INHERIT);
}

void __retarget_lock_close(_LOCK_T lock)
{
	tx_mutex_delete(&lock->Mutex);
	free(lock);
}

void __retarget_lock_close_recursive(_LOCK_T lock)
{
	tx_mutex_delete(&lock->Mutex);
	free(lock);
}

void __retarget_lock_acquire (_LOCK_T lock)
{
	tx_mutex_get(&lock->Mutex,TX_WAIT_FOREVER);
}

void __retarget_lock_acquire_recursive (_LOCK_T lock)
{
	tx_mutex_get(&lock->Mutex,TX_WAIT_FOREVER);
}

int __retarget_lock_try_acquire(_LOCK_T lock)
{
  return tx_mutex_get(&lock->Mutex,TX_NO_WAIT);
}

int __retarget_lock_try_acquire_recursive(_LOCK_T lock)
{
	return tx_mutex_get(&lock->Mutex,TX_NO_WAIT);
}

void __retarget_lock_release (_LOCK_T lock)
{
	tx_mutex_put(&lock->Mutex);
}

void __retarget_lock_release_recursive (_LOCK_T lock)
{
	tx_mutex_put(&lock->Mutex);
}

void _tx_execution_initialize(VOID)
{
	//nop
}

VOID _tx_execution_thread_enter(void)
{
	_impure_ptr = &_tx_thread_current_ptr->newlib;
}

VOID _tx_execution_thread_exit(void)
{
	_impure_ptr = &_impure_data;
}

VOID _tx_execution_isr_enter(void)
{
	tx_execution_isr_nest_counter++;

	if ((TX_THREAD_GET_SYSTEM_STATE()) && (tx_execution_isr_nest_counter == 1))
	{
		_impure_ptr = &_impure_data;
	}
}

VOID _tx_execution_isr_exit(void)
{
	if ((TX_THREAD_GET_SYSTEM_STATE()) && (tx_execution_isr_nest_counter == 1))
	{
		TX_THREAD* thread_ptr = _tx_thread_current_ptr;

		if (thread_ptr)
		{
			_impure_ptr = &thread_ptr->newlib;
		}
	}

	tx_execution_isr_nest_counter--;
}
```

When it's relevent, i added _tx_execution_isr_enter/exit in my isr.
```

void xxx_ISR_Handler(void)
{
	_tx_execution_isr_enter();
//your code
	_tx_execution_isr_exit();
}
```

amgross, 2025-07-27:
Hi @Mula-Gabriel @sanderson-AstroFirmwareSolutions 
1. Aren't we need different _impure_ptr for each interrupt level? (in case it in middle of use and interrupted by higher priority interrupt)
2. What do you do about locks in ISR? Probably need to assert on it if someone calls function that try acquire mutex in ISR? (threadx return error in case it is not TX_NO_WAIT timeout). Need to be careful not to raise assert if acquire called before scheduler and ence failed.

billlamiework, 2025-07-28:
> I have a feeling that the newlib functions that uses the locks and the reent struct are not intended to run from ISR at all.
> Because also in bare metal system there is need to handle those cases and I don't see they are referring to it.
> Hence I think there is no need to handle ISR at all.

Yes, anything that needs a mutex lock cannot be called from an ISR…  doing so should be a system error.

amgross, 2025-07-29:
hi @Mula-Gabriel @sanderson-AstroFirmwareSolutions @billlamiework 
I made some improvements to the lock functions that @Mula-Gabriel showed (ignoring now the reentrance, this is different issue and I don't want to mix). I will be glad if you can review it.
The differences are basically:
1. return code checking (will catch also calling lock from ISR)
2. implementing none recursive mutex from threadx recursive mutex
3. skipping locking if scheduler didn't started yet
/*
 * https://sourceware.org/newlib/libc.html#Reentrancy :
 * 
 * Newlib was configured to allow the target platform to provide the locking routines and static locks at link time.
 * As such, a dummy default implementation of these routines and static locks is provided for single-threaded
 * application to link successfully out of the box on bare-metal systems.
 *
 * For multi-threaded applications the target platform is required to provide an implementation for all these
 * routines and static locks. If some routines or static locks are missing, the link will fail with doubly defined symbols.
 */

```
#include <assert.h>

typedef struct __lock * _LOCK_T;

#include "tx_api.h"
#include "tx_thread.h"

// For smp threadx need to use _tx_thread_system_state[0]
// It is not enough to check if different than 0 because in some ports (ARC EM for example) in interrupts this value may be higher.
#define THREADX_NOT_FINISH_INIT() (_tx_thread_system_state > ((ULONG) 0xF0000000UL))

struct __lock
{
	TX_MUTEX Mutex;
};

struct __lock __lock___sinit_recursive_mutex;
struct __lock __lock___sfp_recursive_mutex;
struct __lock __lock___atexit_recursive_mutex;
struct __lock __lock___at_quick_exit_mutex;
struct __lock __lock___malloc_recursive_mutex;
struct __lock __lock___env_recursive_mutex;
struct __lock __lock___tz_mutex;
struct __lock __lock___dd_hash_mutex;
struct __lock __lock___arc4random_mutex;

// This function should be called from the tx_application_define function
void init_retarget_locks(void)
{
    UINT tx_ret;

    /*
     * In case you have mutex that not expected to be in use, you can but in comment the mutex and its 
     * initialization (if it is linked in, you will get lots of weird linker issues about duplicate definitions)
     */

    tx_ret = tx_mutex_create(&__lock___sinit_recursive_mutex.Mutex,"init",TX_INHERIT);
    assert(tx_ret == TX_SUCCESS);

    tx_ret = tx_mutex_create(&__lock___sfp_recursive_mutex.Mutex,"sfp",TX_INHERIT);
    assert(tx_ret == TX_SUCCESS);

    tx_ret = tx_mutex_create(&__lock___atexit_recursive_mutex.Mutex,"atexit",TX_INHERIT);
    assert(tx_ret == TX_SUCCESS);

    tx_ret = tx_mutex_create(&__lock___at_quick_exit_mutex.Mutex,"aqe",TX_INHERIT);
    assert(tx_ret == TX_SUCCESS);

    tx_ret = tx_mutex_create(&__lock___malloc_recursive_mutex.Mutex,"malloc",TX_INHERIT);
    assert(tx_ret == TX_SUCCESS);

    tx_ret = tx_mutex_create(&__lock___env_recursive_mutex.Mutex,"env",TX_INHERIT);
[... 144 lines cut ...]
    UINT tx_ret;

    if(THREADX_NOT_FINISH_INIT())
    { // Before there is scheduler can't use and no need mutexes
        return;
    }

    tx_ret = tx_mutex_put(&lock->Mutex);
    assert(TX_SUCCESS == tx_ret);
}

void __retarget_lock_release_recursive (_LOCK_T lock)
{
    UINT tx_ret;

    if(THREADX_NOT_FINISH_INIT())
    { // Before there is scheduler can't use and no need mutexes
        return;
    }

    tx_ret = tx_mutex_put(&lock->Mutex);
    assert(TX_SUCCESS == tx_ret);
}

```

amgross, 2025-07-30:
Thanks for the reply @Mula-Gabriel 

>I don’t understand the point of checking whether the OS is running. If the OS isn’t running, locks are pointless: the code that tries to acquire a mutex will simply return an error, the mutex won’t be acquired — so we really don’t care.

I want to differ the error case of init state from other failures, other failures causes assert to be raised (and this failure I want to just ignore).

>As for interrupts, it’s perfectly valid to acquire a mutex from within an ISR. What’s forbidden is waiting for a mutex inside an ISR. This can still be useful to protect a resource in the case of preemptible interrupts. This case is already handled by ThreadX.

As you can see, I didn't made any explicit reference to ISR in my code. I expect locks to fail and raise assert (as they not allowed in ISR in threadx) but try-lock to be OK. This is due to:
https://github.com/eclipse-threadx/threadx/blob/12dac1d67d0dfe1c06f839ee285e190c759865a7/common/src/txe_mutex_get.c#L110-L118

>The purpose of _tx_execution_isr_enter / _tx_execution_isr_exit is to protect Newlib thread context from isr , not to run Newlib within an ISR. So I have one context for ISRs and one per thread — which works perfectly for me. using heavy newlib fonction inside isr should be avoid if it's possible.

I totally ignored currently the reentrance issue in my code, will discuss it seperatly. I don't want to mix the reentrance struct topic with the lock topic.

>I also don’t understand why you’re adding assert() calls in a "try" function. Isn’t it completely normal for a try to fail ?

It is normal to fail just if the reason is that someone else aquired the mutex (not NULL pointer for example). You can see in my `try` functions that before I assert on failure I check if failed due to `TX_NOT_AVAILABLE` and if this is the case I just return error, not assert. any other failure go into assert.

>And finally, I really don’t get why you’re trying to convert a recursive mutex into a simple mutex — especially since you don’t control how these mutexes are acquired or released, as that’s handled by Newlib. I think you're overcomplicating things for no real reason.

If newlib want non-recursive mutex then I want to give it exactly this. I don't want to assume that newlib can work with recursive although it requests from me non-recursive, because even if now this assumption is correct, it may change one day.

billlamiework, 2025-07-30:
ThreadX does not expect mutex calls from an ISR or before initialization is complete. It also expects interrupts to be disabled until tx_application_define returns.  Just fyi…

https://github.com/eclipse-threadx/threadx/blob/12dac1d67d0dfe1c06f839ee285e190c759865a7/common/src/txe_mutex_get.c#L110-L118

amgross, 2025-07-30:
Hi @billlamiework 
From code it seems valid to acquire mutex in isr/initialisation with TX_NO_WAIT option
https://github.com/eclipse-threadx/threadx/blob/12dac1d67d0dfe1c06f839ee285e190c759865a7/common/src/txe_mutex_get.c#L110-L118
It also make sense, as the reason to not use it is because ISR and initialisation can't be preempted, but with TX_NO_WAIT it won't get preempted

Mula-Gabriel, 2025-07-30:
@amgross,

under isr, you should exit with TX_CALLER_ERROR
```
        /* Check for interrupt call.  */
        if (TX_THREAD_GET_SYSTEM_STATE() != ((ULONG) 0))
        {

            /* Now, make sure the call is from an interrupt and not initialization.  */
            if (TX_THREAD_GET_SYSTEM_STATE() < TX_INITIALIZE_IN_PROGRESS)
            {

                /* Yes, invalid caller of this function, return appropriate error code.  */
                status =  TX_CALLER_ERROR;
            }
        }
```
i think you could call the _tx_mutex_get function, but it will do nothing.

anyway, the mutex need the calling thread for owner pointer, wich you can't obtain properly from isr...
(from the isr you will get the last exe thread...)

i guess this is the root cause for [billlamiework](https://github.com/billlamiework) saying "can't use mutex in isr"

see in **_tx_mutex_get**
```
 /* Pickup thread pointer.  */
    TX_THREAD_GET_CURRENT(thread_ptr)

    /* Determine if this mutex is available.  */
    if (mutex_ptr -> tx_mutex_ownership_count == ((UINT) 0))
    {

        /* Set the ownership count to 1.  */
        mutex_ptr -> tx_mutex_ownership_count =  ((UINT) 1);

        /* Remember that the calling thread owns the mutex.  */
        **mutex_ptr -> tx_mutex_owner =  thread_ptr;**
```

amgross, 2025-07-30:
Oops @Mula-Gabriel @billlamiework 
You are both correct, I missed the check @Mula-Gabriel referred to.
I will fix the relevant code and send the fixed points.

fdesbiens, 2026-09-23:
I opened eclipse-threadx/rtos-docs-asciidoc#78 to track documentation of the current ThreadX/Newlib usage and limitations. I also opened eclipse-threadx/samplex#73 with the `help wanted` label because we need help building and validating a working sample. I am closing this issue because the original question was answered above and the remaining work has dedicated tickets.

closed 2026-09-23 by fdesbiens