eclipse-threadx/threadx #450: txm_module_manager_absolute_load create wrong data start pointer

amgross, 2025-05-14:
**Describe the bug**
In _txm_module_manager_absolute_load There is:
https://github.com/eclipse-threadx/threadx/blob/7ad78c40e9702917264a9b722890110fdb909ebc/common_modules/module_manager/src/txm_module_manager_absolute_load.c#L314
This causes the txm_module_instance_data_start to be defined as size of struct + size of code, which not make sense to me, as it not address (and this variable should be address, as it used for things like MPU and checking if things located in memory area).

**Expected behavior**
txm_module_instance_data_start should contain the module data memory pointer

**Impact**
In edge cases like usage of MPU or objects that located in this memory things may fail

fdesbiens, 2025-05-20:
Thank you for the bug report @amgross. We will discuss it.

amgross, 2025-05-21:
I would just want to add about my MPU point, that probably (although not checked) the current MPU code won't work in current absolute load.
This is because in different than the other loads here the main data is separated from the start stop+callback threads stack (main data is in absolute place and stacks allocated on byte pool in runtime).
I would consider make it similar to all other loads and but the stacks in the beginning of the (absolute located) main data

fdesbiens, 2025-06-02:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: Please provide your feedback.

fdesbiens, 2026-09-08:
Thank you for the detailed report @amgross, and apologies for the long delay.
Both of your points are confirmed, and PR #699 addresses them.
On the first point, the root cause turned out to be a design gap rather than a miscalculation. An absolutely located module has its code and its data placed at two independent fixed addresses by the module's linker script — in the Cortex-A35 sample these are `__FLASH_segment_start__` and `__RAM_segment_start__`. The module preamble emits only `__code_size__` and `__data_size__`, never the RAM base address, and `txm_module_manager_absolute_load()` receives only the code address. The Module Manager therefore had no way to derive the data area address, so the expression you flagged could not be corrected in place.
The fix adds `txm_module_manager_absolute_load_extended()`, which takes the module's data area address as a fourth argument. `txm_module_manager_absolute_load()` becomes a deprecated wrapper that keeps working for existing callers, following the same deprecation pattern we already use for `txm_module_object_pointer_get()`.
Your MPU observation was well founded. With the data area unknown, `_txm_module_manager_mm_register_setup()` would program a protection region from an invalid base at load time, so the deprecated path now rejects modules that request `TXM_MODULE_MEMORY_PROTECTION` with `TXM_MODULE_INVALID_PROPERTIES` rather than silently misconfiguring the hardware. Using the extended service with the real data address gives you correct MPU coverage.
On your second point, `txm_module_instance_module_data_base_address` was indeed wrong. It pointed one past the end of the byte pool allocation, since for an absolute load that allocation holds only the two default thread stacks. It now points at the module's absolutely located data area.
The documentation is updated in eclipse-threadx/rtos-docs-asciidoc#58.

fdesbiens, 2026-09-08:
PR #699 has been merged into `dev`. `txm_module_manager_absolute_load_extended()` is now available, `txm_module_manager_absolute_load()` is deprecated, and `txm_module_instance_module_data_base_address` is corrected.
The companion documentation PR eclipse-threadx/rtos-docs-asciidoc#58 is still in review and will be merged separately.
Thank you again for the report and for the follow-up analysis, @amgross! Closing this one.

closed 2026-09-08 by fdesbiens