eclipse-threadx/threadx #461: Missing #

42Bastian, 2025-08-15:
**Describe the bug**
This line misses a "#"

https://github.com/eclipse-threadx/threadx/blob/12dac1d67d0dfe1c06f839ee285e190c759865a7/ports/cortex_m33/gnu/src/tx_thread_schedule.S#L145

fdesbiens, 2025-09-02:
Thank you, @42Bastian.

I will discuss this with the team and fix as needed.

billlamiework, 2025-09-15:
Yes, this line should have a #, however, I tried it with one GCC compiler on my side and it generated the correct code.  Are you seeing an actual error on your side?  If so, which GCC compiler version are you using?

42Bastian, 2025-09-16:
GCC (or AS) and as I see IAR:
https://github.com/eclipse-threadx/threadx/blob/12dac1d67d0dfe1c06f839ee285e190c759865a7/ports/cortex_m33/iar/src/tx_thread_schedule.s#L148

do not blame, but for example TI Clang will.

fdesbiens, 2026-09-03:
Thanks for the report @42Bastian, and for confirming it @billlamiework.

I have opened #693 to fix this. While looking into it I found the same missing `#` in 31 places across 22 files, not just the one line you linked: the Cortex-M3, M4, M7, M33, M52, M55 and M85 ports, their module manager counterparts, and the shared ARMv7-M and ARMv8-M sources. The GNU and AC6 variants had already been corrected, so what remained were the IAR and AC5 ones.

I confirmed that both spellings assemble to the same Thumb-2 encoding (`f04f 0000`), so nothing changes in the generated code — it only stops the stricter assemblers from rejecting the source, which was exactly your point.

fdesbiens, 2026-09-23:
I confirmed that #693 fixed the missing immediate marker in `dev`, including the other affected Cortex-M scheduler sources. The fix has not reached `master` yet. I’m closing this issue because the correction has landed in the development branch.

closed 2026-09-23 by fdesbiens